JohnDMcMaster/usbrply #8: Add requirements.txt

JohnDMcMaster, 2019-09-17:
```
import pcap
import argparse
import sys
import binascii
import struct
from collections import namedtuple
import os
import errno
import json
```

JohnDMcMaster, 2019-09-17:
sudo pip install pycap

SteadBytes, 2020-02-29:
I'm having trouble installing this and/or it's dependencies. I have both `pycap` and `python-libpcap` installed however `usbrply` errors with: `ImportError: No module named pcap`. Do you have a known working install that one could dump the `pip` package versions from?

JohnDMcMaster, 2020-03-14:
Collected some info from ubuntu 16 install that probably works (python usbrply.py -h works at least)

```
apt-cache show python-libpcap
Package: python-libpcap
Priority: optional
Section: universe/python
Installed-Size: 139
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian QA Group <packages@qa.debian.org>
Architecture: amd64
Version: 0.6.4-1
Provides: python2.7-libpcap
Depends: python (>= 2.7.1-0ubuntu2), python (<< 2.8), libc6 (>= 2.14), libpcap0.8 (>= 0.9.8)
Conflicts: python-pypcap
Filename: pool/universe/p/python-libpcap/python-libpcap_0.6.4-1_amd64.deb
Size: 34758
MD5sum: 3eaa5cc919f3f3a1d5fefd5a0b9f94e6
SHA1: 0f8ba4542255e9a8cd2fa217feba616d19c1ddac
SHA256: e2a575511681e06b491e95eb6a068444db056a58b39b78568cea5e529031f1d6
Description-en: Python wrapper for libpcap packet capture library
 Python module for the libpcap, the UNIX packet capture library.
 This can be used for many purposes including network debugging,
 traffic analysis, intrusion detection, based on the original
 Python libpcap module by Aaron Rhodes.
Description-md5: 22835ae063f091a6dbac281609504857
Homepage: http://sourceforge.net/projects/pylibpcap
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Origin: Ubuntu
```

It looks like the pylibpcap package doesn't install cleanly though:

```
sudo pip install pylibpcap
...
Collecting pylibpcap
  Could not find a version that satisfies the requirement pylibpcap (from versions: )
No matching distribution found for pylibpcap
...
```

I'm unclear if this is a pip or package issue.

Other: I'm not attached to this pcap lib if its not a good choice

JohnDMcMaster, 2020-03-15:
@SteadBytes FYI significant discussion of the library has popped up in https://github.com/JohnDMcMaster/usbrply/issues/9

JohnDMcMaster, 2020-07-12:
fixed in 136c846

closed 2020-07-12 by JohnDMcMaster