JohnDMcMaster/usbrply #12: TypeError: b2a_hex() argument 1 must be string or buffer, not None

JohnDMcMaster, 2020-01-05:
```
usbrply --packet-numbers --no-setup --device-hi  -j 2019-12-26_07_snap.pcapng 

Traceback (most recent call last):
  File "/home/mcmaster/bin/usbrply", line 1172, in <module>
    gen.run()
  File "/home/mcmaster/bin/usbrply", line 613, in run
    p.loop(-1, self.loop_cb)
  File "/usr/lib/python2.7/dist-packages/pcap.py", line 115, in loop
    def loop(self, *args): return _pcap.pcapObject_loop(self, *args)
  File "/home/mcmaster/bin/usbrply", line 768, in loop_cb
    self.process_complete(dat_cur)
  File "/home/mcmaster/bin/usbrply", line 823, in process_complete
    self.processBulkComplete(dat_cur)
  File "/home/mcmaster/bin/usbrply", line 1113, in processBulkComplete
    self.processBulkCompleteOut(dat_cur)
  File "/home/mcmaster/bin/usbrply", line 1096, in processBulkCompleteOut
    'data': bytes2AnonArray(self.submit.m_data_out),
  File "/home/mcmaster/bin/usbrply", line 290, in bytes2AnonArray
    return binascii.hexlify(bytes)
TypeError: b2a_hex() argument 1 must be string or buffer, not None
```

https://drive.google.com/file/d/135_c2xcOkmGNMkW_2O4TxbmhdThg_k4Q/view?usp=sharing

WaseemAlkurdi, 2020-07-07:
@JohnDMcMaster Any progress on this?

JohnDMcMaster, 2020-07-07:
No, but sounds like its affecting people so I'll look into it

WaseemAlkurdi, 2020-07-07:
> No, but sounds like its affecting people so I'll look into it

First, it's amazing how fast you replied!
Another thing that had me confused is that there are two incompatible `pcap` libraries for GitHub: `pypcap` and `pylibpcap` ... and it seems that Debian and Ubuntu are only packaging the former these days.
Building `pylibpcap` from source results in an odd (but easily worked around) C++ issue (I'll recreate it soon, should you need the precise error text).
Then, `pylibpcap` would install, but `usbrply` itself won't, throwing this error message. It would really be nice if you look into this!

JohnDMcMaster, 2020-07-07:
np! its hit or miss you just happened to hit me at a time when I'm procrastinating starting a consulting project heh

I've created a ticket, hopefully this helps clarify the current issues: https://github.com/JohnDMcMaster/usbrply/issues/36

JohnDMcMaster, 2020-07-12:
Fixed in 22489bf559ab8fae335adf5cda39c5873062afe9

Looks like it was caused by incorrectly handling recycled URBs which are partially received (ie we get the complete but not the submit at the beginning of the capture). Linux based captures should have primarily been affected although Windows possibly was as well.

Please let me know if you still have issues!

closed 2020-07-12 by JohnDMcMaster