JohnDMcMaster/usbrply #25: Bring back C output targets

JohnDMcMaster, 2020-04-01:
These were removed during the recent rework. But pyprinter.py is pretty straightforward, and it should be easy to replicate that as a template to bring them back

himekifee, 2022-01-22:
Any chance to get this back? Kinda looking for it rn.

JohnDMcMaster, 2022-04-23:
@himekifee not sure if this is still of use to you, but I have a WIP: https://github.com/JohnDMcMaster/usbrply/pull/74

Did you want kernel or libusb?

himekifee, 2022-04-23:
Libusb is generally enough for me. I used to prototype in libusb and if necessary, port to kernel code.

JohnDMcMaster, 2022-04-24:
I've added experimental support but its pretty untested (compiles but not tested on real device). If you happen to test it definitely interested in your feedback, otherwise I'll test it properly when I can