JohnDMcMaster/usbrply #26: Add timestamps to JSON

JohnDMcMaster, 2020-04-01:
The ability to making timing accurate replays (ie insert sleeps) was lost as part of recent rework

krasin, 2020-04-01:
It was something missing before the recent rework, I believe. At some point, I discovered that in a hard way, when galvo head and laser were acting without a proper sync which resulted in random cuts across the table. :)

Would be nice to get it back.

JohnDMcMaster, 2020-04-01:
Heh oops!

It wasn't in the JSON, but it was in direct python output if you did --sleep:
```
  --sleep               Insert sleep statements between packets to keep
                        original timing
```

--sleep is still there but the corresponding logic is commented out: https://github.com/JohnDMcMaster/usbrply/blob/master/usbrply/pyprinter.py#L133

krasin, 2020-04-01:
Ah, cool! In this case, it's just me not reading the docs. Sorry about that. :)

JohnDMcMaster, 2020-07-12:
Fixed for Linux in 2963b62. Windows URB doesn't include timestamp so I don't currently have any metadata to go off of. Note this is a library issue, the actual data is there its just that I can't get to it. Hopefully this will get fixed in some way as part of https://github.com/JohnDMcMaster/usbrply/issues/36

closed 2020-07-12 by JohnDMcMaster