JohnDMcMaster/usbrply #30: Can replay only on Linux?

stjeong, 2020-04-14:
I didn't see any words about that. Can replay on Windows?

JohnDMcMaster, 2020-04-14:
Hi! Thanks for reaching out. usbrply probably runs on Widows, but as I'm not a Windows user and do not know much about Python there I cannot comment. If you try things though, I'd love to hear feedback and/or instructions.

kafisatz, 2020-07-11:
Hi there.
I wanted to try this under windows.

It is not clear to me how I should run this step 'Linux: run: "usbrply --device-hi -p my.pcapng >replay.py"'
can you elaborate please?

JohnDMcMaster, 2020-07-12:
Hi kafisatz,
Unfortunately there haven't been any major changes since my previous comment. I'm not a Windows person and haven't had time to research this. Current effort will be put into fixing some of the bugs affecting the framework in general. Some of this may include things like upgrading the packet library / Python3 which may be required for Windows anyway.

JohnDMcMaster, 2020-07-12:
I've cut a pip release and added Windows instructions (see latest README). Probably could be more streamlined but this should work. Please let me know if you run into any issues

closed 2020-07-12 by JohnDMcMaster