JohnDMcMaster/usbrply #32: No module named usbrply.parsers

kouglov, 2020-04-20:
Hi, I've been struggling with making all of this work for a few days.

Been launching setup.py to build & install.
Then when launching with a pcapng file, got this error on line 4, when importing : 

ImportError : no module named usbrply.parsers

But the parser python file is present in the folder, did I builded something wrong ?

Maybe this is evident for python coders, but for example, the main.py script should be renamed usbrply.py, I've also been struggling with this for a few minutes before understanding it had to be renamed. (this also crash the pip install)

JohnDMcMaster, 2020-05-19:
Oh sorry looks like I missed this... Anyway it looks like setup.py was broken. Can you retry?

sannong, 2020-05-19:
Hello,
I just tried this with the latest commit from 11 hours ago (1e60c33) and setup.py runs fine and installs but I also get the "ImportError: No module named usbrply.parsers" error when I try and run. The full trace is: 

`usbrply --device-hi -p test.pcapng >replay.py
Traceback (most recent call last):
  File "/usr/local/bin/usbrply", line 4, in <module>
    __import__('pkg_resources').run_script('usbrply==1.0.0', 'usbrply')
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1469, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/usbrply-1.0.0-py2.7.egg/EGG-INFO/scripts/usbrply", line 4, in <module>
    __import__('pkg_resources').run_script('usbrply==1.0.0', 'usbrply')
ImportError: No module named usbrply.parsers`

leohearts, 2020-05-30:
I met the same problem.Can you fix it?

krasin, 2020-05-30:
@leohearts I understand that you probably didn't mean it, but just pointing out to the fact that your question could be interpreted as pretty rude to the author of this library, as it has an implicit expectation that he owes you a fix.

leohearts, 2020-05-30:
@krasin Sorry, i just mean i met the same problem, nothing more.
I'm not good at using English, and i don't know the emotion in texts, sorry for that.

krasin, 2020-05-30:
No worries, I am not good at English too. :)

While I am not the author of the library, I will try to reproduce this issue to see if there's an obvious fix.

JohnDMcMaster, 2020-05-30:
Heyo,
I've been a bit busy with work recently, but I actually have a bit of a break right now. I just took a quick look and its a regression from when I moved to a library (ie the library isn't being added to the dist). I'll try to have a patch soon

krasin, 2020-05-30:
@leohearts can you please test if my proposed fix addresses your issue? It would be nice to hear, if it works for you.

closed 2020-05-30 by JohnDMcMaster

leohearts, 2020-05-30:
@krasin It works now! Thank you!

krasin, 2020-05-30:
@leohearts you're welcome! I am glad it worked.

PR #34 (2020-05-30): setup.py: don't forget to install packages.
Hi John,

this is a lame attempt to fix #32. I don't have full confidence that it does the trick, because my system is Python3 and the mainline usbrply is broken for me due to #9. But I have some hope that my patch makes it better.

Please, don't merge until somebody confirms that it resolved the issue for them.
commit dfef9cb2 (2020-05-30): setup.py: don't forget to install packages.

Hopefully, fixes #32.