JohnDMcMaster/usbrply #33: No usbrply.py?

sannong, 2020-05-18:
Hello, 

I just downloaded the code (commit 9bf1beafaff129fb0c9ad6ad3665de07ce8f9e73) and am attempting to get it setup in my environment. The readme indicates to run 'usbrply --device-hi -p my.pcapng >replay.py' but there is no usbrply executable or file to call in any directory? Similarly setup.py searches for usbrply.py but cannot find it. Is there some other build step needed not specified in the readme that needs to be performed prior to use?

Thank you

JohnDMcMaster, 2020-05-19:
Hi there! It looks like there was possibly a regression during a recent major refactor. set.up and/or the instructions probably need to be updated. I'll get back to you shortly

JohnDMcMaster, 2020-05-19:
Should be fixed in master (via https://github.com/JohnDMcMaster/usbrply/commit/cdea59e78becb6950f7e36ed32c2ae7057ba4822)

Please let me know if you have any more issues!

closed 2020-05-19 by JohnDMcMaster

sannong, 2020-05-19:
Thanks! Works now.