JohnDMcMaster/usbrply #36: Easier to setup pcap library

JohnDMcMaster, 2020-07-07:
TLDR: the current library we use is very fast and works well but is python2 only. Now that parsing packet files is separated from generating output we should add more parsers.

In order to make this work we also need to clean up and solidify the .json intermediate representation (IR). There is a lot of churn on the IR for higher level parsing that could in theory effect this (ex: for FTDI/MPSSE support) but can probably use its own file format that's loosely related to the core IR. In particular I'd like packet parsing to be capable of streaming (more of an API than a format issue), but this is currently not possible due to how metadata is added

Following this it should be possible to cleanly add more packet parsing libraries. IIRC there maybe even was a PoC on the python 3 ticket

See also the python 3 ticket which has a lot of discussion on this, but really should have a dedicated but related ticket: https://github.com/JohnDMcMaster/usbrply/issues/9

kafisatz, 2020-07-12:
``sudo apt-get install -y python-libpcap```
does not work for me on Ubuntu 20.04
it says unable to locate package. 
Any ideas why?

JohnDMcMaster, 2020-07-12:
kafisatz: try thew library. If you are still have issues please open a new ticket.

```
sudo pip install python-pcapng
```

5017eee Adds the pcapng library. By default it searches for the faster pcap library, but defaults to pcapng if that isn't available (also possibly erroring if that isn't available)

closed 2020-07-12 by JohnDMcMaster