JohnDMcMaster/usbrply #40: Use timestamps from loop callback, not urb

JohnDMcMaster, 2020-07-12:
JSON has some timestamps (ex: for --sleep) but they are currently calculated from the Linux URBs which isn't portable. Instead calculate from the packet timestmap

worldjoe, 2021-11-25:
Would this fix the issue I am having? I have a USB device with LEDs on it. I use the USBpcap to capture all the packets. The application the drives the USB device (during capture) must be timing the USB messages sent over, because the entire LED show takes about 10 minutes. But when I replay the packet, the entire show is done in a flash. It's clear that the intro and extro are working properly, but in the middle it's all noise with all the LEDs on at once essentially. I suspect that the python replay isn't keyed to the same timing that the capture was.

worldjoe, 2021-11-27:
Oh never mind. I looked at the json output generated by passing the -j flag and I don't see any timestamp data in on my packets. I was hoping there would be timestamps in the data so that I could attempt to replay them at the same rate/time as they were originally sent.

Here is the -j output for a single request. But maybe there is more data in the pcappng file that isn't being outputted in the json format. I'll need to investigate the original file.
`{
            "bRequest": 16,
            "bRequestType": 192,
            "comments": [
                "Generated from packet 90291/90293"
            ],
            "complete": {
                "packn": 90293,
                "urb": {
                    "bus_id": 1,
                    "data_length": 19,
                    "device": 13,
                    "endpoint": 128,
                    "id": 18446666073525668464,
                    "irp_info": 1,
                    "irp_status": 0,
                    "pcap_hdr_len": 28,
                    "transfer_type": 2,
                    "usb_func": 8
                }
            },
            "data": "00000000000000000000000000000000000000",
            "submit": {
                "packn": 90291,
                "urb": {
                    "bus_id": 1,
                    "data_length": 8,
                    "device": 13,
                    "endpoint": 128,
                    "id": 18446666073525668464,
                    "irp_info": 0,
                    "irp_status": 0,
                    "pcap_hdr_len": 28,
                    "transfer_type": 2,
                    "usb_func": 8
                }
            },
            "type": "controlRead",
            "wIndex": 0,
            "wLength": 19,
            "wValue": 0
        }`

JohnDMcMaster, 2021-11-28:
Did you capture the packets in Windows or Linux? If you captured them on Windows, then yes you would be affected by this

JohnDMcMaster, 2021-11-30:
I have an experimental fix ready: https://github.com/JohnDMcMaster/usbrply/pull/58

closed 2022-03-13 by JohnDMcMaster