JohnDMcMaster/usbrply #41: Python script without data

QkiZMR, 2020-08-13:
I made some captures on Windows with Wireshark and usbpcap and then saved it to file.
[profil2.zip](https://github.com/JohnDMcMaster/usbrply/files/5071063/profil2.zip)
Then I tried convert it python script. usbrply generated script but without any data in it. I made few others captures and tried to convert them but all always converted to same script without any data.
[profil2.py.zip](https://github.com/JohnDMcMaster/usbrply/files/5071078/profil2.py.zip)

JohnDMcMaster, 2020-08-14:
Interrupt support is pretty poor and had been explicitly filtered out. I just made a quick stab to support your data. Please let me know if this helps.

QkiZMR, 2020-08-16:
I updated local repo and made `sudo python setup install` again. When I'm trying to run `usbrply` I'm getting following error:

```
 $ usbrply --device-hi -p profil1.pcapng > profil1.py
Traceback (most recent call last):
  File "/usr/local/bin/usbrply", line 4, in <module>
    __import__('pkg_resources').run_script('usbrply==2.0.1', 'usbrply')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 666, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1469, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/usbrply-2.0.1-py2.7.egg/EGG-INFO/scripts/usbrply", line 4, in <module>
    __import__('pkg_resources').run_script('usbrply==2.0.1', 'usbrply')
  File "build/bdist.linux-x86_64/egg/usbrply/parsers.py", line 6, in <module>
  File "/usr/local/lib/python2.7/dist-packages/usbrply-2.0.1-py2.7.egg/usbrply/win_pcap.py", line 494
    self.verbose and print("WARNING: unknown transfer type %u" % self.urb.transfer_type)
                         ^
SyntaxError: invalid syntax
```

JohnDMcMaster, 2020-08-16:
ah yeah py2 has been on the way out but forgot to update the default. Fixed to default to py3

Hmm unfortunately not sure what I can do to help further. I'd suggest taking a look at wireshark GUI vs the generated python code to see if it looks reasonable?

Also: did you include --wrapper? Otherwise it won't give you a main() to actually execute the code

JohnDMcMaster, 2020-10-15:
Please let me know if there is anything I can do, otherwise I will assume this is solved and close ticket for inactivity. Thank you.

JohnDMcMaster, 2020-12-24:
Closed for inactivity.

closed 2020-12-24 by JohnDMcMaster