JohnDMcMaster/usbrply #43: How to play back USB traffic?

kelkka, 2020-09-23:
Hello,

Maybe I don't follow exactly what this software does, but I'm looking to record USB-traffic on Windows and be able to playback that exact traffic later.

So, I record traffic using wireshark and usbpcap, and I try moving the mouse which is device 3.9.1 and expect that the mouse movement will replay if I run the script output by usbrply later.

I open the pcapng with usbrply with:
usbrply --device-hi -p mousecap.pcapng

which prints:
`
# Generated by usbrply
# cmd: usbrply --device-hi -p mousecap.pcapng

def replay(dev):
    def bulkRead(endpoint, length, timeout=None):
        return dev.bulkRead(endpoint, length, timeout=(1000 if timeout is None else timeout))

    def bulkWrite(endpoint, data, timeout=None):
        dev.bulkWrite(endpoint, data, timeout=(1000 if timeout is None else timeout))
    
    def controlRead(request_type, request, value, index, length,
                    timeout=None):
        return dev.controlRead(request_type, request, value, index, length,
                    timeout=(1000 if timeout is None else timeout))

    def controlWrite(request_type, request, value, index, data,
                     timeout=None):
        dev.controlWrite(request_type, request, value, index, data,
                     timeout=(1000 if timeout is None else timeout))

    # Generated by usbrply
    # Source: Windows pcap (USBPcap)
    # cmd: usbrply --device-hi -p mousecap.pcapng
    # Selected device 14
`

I tried using --device 9 to maybe make it use the mouse? But it says it selected device 14 anyway and nothing changes. I managed to get Selected device 9 at some point but it doesnt seem to matter (should really be 3.9.1 anyway? But dots are not allowed when specifying device). The capture file has a bunch of data but the script has nothing except the default text as I understand it.

Not sure what's going on or what I should expect from this software. Are there only specific devices that are supported? How do I know if my USB-device is supported in that case?

Thanks

JohnDMcMaster, 2020-09-25:
Hi! I'm guessing this is HID traffic which is interrupts / isochronous. Generally usbrply doesn't support that terribly well. However if you send you capture over I can look into adding better support.

JohnDMcMaster, 2020-10-15:
Please let me know if there is anything I can do, otherwise I will assume this is solved and close ticket for inactivity. Thank you.

kelkka, 2020-10-21:
Sorry, I had to put this on hold for a while, and I no longer have the recording, but you can record this easily yourself by just using any USB mouse I guess, yes it was HID. 
I think I'll have to use some other software for my purpose anyway, the mouse was just a test to see how/if this one works.

closed 2020-10-21 by kelkka