JohnDMcMaster/usbrply #45: No data generated, just "WARNING: Packet X missing submit."

nametable, 2020-12-21:
When I try to create a replay of a USB-FAN device, I just get a bunch of "WARNING: Packet X missing submit." The capture file is included below in the zip:
[capture.zip](https://github.com/JohnDMcMaster/usbrply/files/5725925/capture.zip)

This is some of the output in `replay()`

```python
    # Generated by usbrply
    # Source: Windows pcap (USBPcap)
    # cmd: ../../../usbrply/main.py --wrapper --device 7160 -p Verylong_USB-FAN_22-07-2019.pcapng
    # Selected device 2
    # WARNING: Packet 4 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 8 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 12 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 16 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 20 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 24 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 28 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 32 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 36 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 40 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 44 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 48 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 52 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 56 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 60 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 64 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 68 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 72 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 76 missing submit.  URB ID: 0xFFFFA48949A7E050
    # WARNING: Packet 80 missing submit.  URB ID: 0xFFFFA48949BB9AF0
    # WARNING: Packet 84 missing submit.  URB ID: 0xFFFFA48949A7E050
```

JohnDMcMaster, 2020-12-23:
H! TLDR: Please 1) update to newest git 2) remove --device and let me know if this fixes your issue

Details:
Looks like there are two issues:
-Don't specify "--device". By default it will select the highest numbered device (in this case "2", not "7160")
-Your kernel is generating a different data flow than I tested. I've updated code to reflect it

nametable, 2020-12-23:
I now see that the generated script is not all warnings anymore. However, I am having an issue where I get a usb error when I try to run the generated script (I can create a different github issue for this if you want). Here's  what the error looks like though:
```Found device
Bus 001 Device 021: ID 0c45:7160
Traceback (most recent call last):
  File "/tmp/USB-Fan/Wireshark captures/Data send/long.py", line 2450, in <module>
    dev.claimInterface(0)
  File "/home/logan/.local/lib/python3.9/site-packages/usb1/__init__.py", line 1168, in claimInterface
    mayRaiseUSBError(
  File "/home/logan/.local/lib/python3.9/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/home/logan/.local/lib/python3.9/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorBusy: LIBUSB_ERROR_BUSY [-6]
```

Maybe this just indicates that the usb device is not setup correctly (udev settings or permissions) but I'm not really sure...

JohnDMcMaster, 2020-12-23:
No it's fine to continue here

Can you try adding "sudo" to the front of your replay command? This might be a permissions issue

nametable, 2020-12-23:
No change when I run `sudo python long.py`

nametable, 2020-12-23:
I think maybe the device is being claimed by Linux as a HID. Maybe I can blacklist the device or "unclaim" it

nametable, 2020-12-23:
I tried something else and got a different error. Under the idea that somehow the USB device was claimed, I did the following:
```bash
sudo su root
cd /sys/bus/usb/drivers/usbhid
echo -n "1-5:1.0" > unbind
```
It now seems to get further, making it to the first `controlWrite` before failing:
```
sudo python long.py 
Scanning for devices...

Found device
Bus 001 Device 029: ID 0c45:7160
Traceback (most recent call last):
  File "/tmp/USB-Fan/Wireshark captures/Data send/long.py", line 2452, in <module>
    replay(dev)
  File "/tmp/USB-Fan/Wireshark captures/Data send/long.py", line 46, in replay
    controlWrite(0x40, 0x40, 0x2A06, 0x0017, b"")
  File "/tmp/USB-Fan/Wireshark captures/Data send/long.py", line 31, in controlWrite
    dev.controlWrite(bRequestType, bRequest, wValue, wIndex, data,
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 1349, in controlWrite
    return self._controlTransfer(request_type, request, value, index, data,
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 1326, in _controlTransfer
    mayRaiseUSBError(result)
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorPipe: LIBUSB_ERROR_PIPE [-9]```

JohnDMcMaster, 2020-12-23:
Hmm IIRC this usually means the device didn't like the data sent to it. Some suggestions:
-Verify you didn't miss an initialization sequence, such as when the device is first plugged in
-Check the USB output in say Wireshark reasonably matches the python code
-Capture Windows traffic and compare to Linux traffic. Ideally as a Windows VM such that both use same Linux usbmon interface and so can be directly compared

nametable, 2020-12-23:
I will give that a try tomorrow. I think there was some captures of the USB
device connecting that I didn't take into account.

nametable, 2020-12-23:
Okay, so I tried monitoring usb traffic to a Windows vm, which was successful using Wireshark. I was able to record everything from connect to data transfer. However, in this new capture I now have a different problem which seems quite unrelated. For reference I captured traffic with usbmon on Archlinux with Wireshark Version 3.4.2. I started capture right before telling `virt-manager` to redirect usb to the vm.

Here's the new error I'm getting:
```
$ python ../../../usbrply/main.py --wrapper -p logan_cap.pcapng > doit.py
Traceback (most recent call last):
  File "/tmp/USB-Fan/Wireshark captures/Data send/../../../usbrply/main.py", line 131, in <module>
    main()
  File "/tmp/USB-Fan/Wireshark captures/Data send/../../../usbrply/main.py", line 127, in main
    usbrply.printers.run(args.ofmt, filtered, argsj=argsj)
  File "/tmp/usbrply/usbrply/printers.py", line 14, in run
    printer.run(j)
  File "/tmp/usbrply/usbrply/pyprinter.py", line 229, in run
    for d in j["data"]:
  File "/tmp/usbrply/usbrply/setup_filter.py", line 31, in gen_data
    for data in datas:
  File "/tmp/usbrply/usbrply/com_pcap.py", line 75, in gen_data
    if not parser.next(self.loop_cb):
  File "/tmp/usbrply/usbrply/pcap_util.py", line 84, in next
    loop_cb(block.captured_len, block.packet_data, block.timestamp)
  File "/tmp/usbrply/usbrply/lin_pcap.py", line 322, in loop_cb
    assert len(self.pcomments) == 0, ("unhandled comment", self.pcomments)
AssertionError: ('unhandled comment', ['Generated from packet 67/74'])
```
Here is my new .pcapng zipped up:
[logan_cap.zip](https://github.com/JohnDMcMaster/usbrply/files/5736709/logan_cap.zip)

JohnDMcMaster, 2020-12-24:
This looks like https://github.com/JohnDMcMaster/usbrply/issues/42

I'll take a quick stab at it in the near future though using your data as a reference

JohnDMcMaster, 2020-12-24:
I've made a quick effort to support Linux interrupts. Please update and let me know if it helps. Sample output:

```
# Generated from packet 67/74
buff = interruptRead(0x81, 0x0002)
validate_read(b"\x24\x80", buff, "packet 67/74")
```

nametable, 2020-12-24:
Yes, it definitely helped! No errors during python script generation. After unbinding my device to avoid the `usb1.USBErrorBusy: LIBUSB_ERROR_BUSY [-6]` error, I was able to run the generated script, which did what was expected on the target device! One of the devices LED's blinked indicating the LED FAN was being programmed.

Thanks for all your help (and very quick too)!

JohnDMcMaster, 2020-12-24:
Great to hear it's working! Glad to be of help. Closing the issue as it seems this is resolved

closed 2020-12-24 by JohnDMcMaster