JohnDMcMaster/usbrply #48: Failed to identify packet format

durka, 2021-10-01:
usbrply won't touch this capture that I recorded using Wireshark on Linux using usbmon.

```
$ usbrply -p /tmp/ver.pcapng 
Traceback (most recent call last):
  File "/usr/local/bin/usbrply", line 4, in <module>
    __import__('pkg_resources').run_script('usbrply==2.0.1', 'usbrply')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 665, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1470, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/EGG-INFO/scripts/usbrply", line 124, in <module>
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/EGG-INFO/scripts/usbrply", line 113, in main
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/usbrply/parsers.py", line 19, in pcap2json
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/usbrply/pcap_util.py", line 137, in guess_parser
AssertionError: failed to identify packet format
```

In order to minimize the example this pcapng contains a single packet exported from Wireshark (had to rename for github):
[ver.pcapng.txt](https://github.com/JohnDMcMaster/usbrply/files/7263586/ver.pcapng.txt)

durka, 2021-10-01:
I tried the same thing exporting to pcap instead and it won't touch that either.

```
$ usbrply -p /tmp/ver.pcap
Traceback (most recent call last):
  File "/usr/local/bin/usbrply", line 4, in <module>
    __import__('pkg_resources').run_script('usbrply==2.0.1', 'usbrply')
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 665, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1470, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/EGG-INFO/scripts/usbrply", line 124, in <module>
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/EGG-INFO/scripts/usbrply", line 113, in main
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/usbrply/parsers.py", line 19, in pcap2json
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/usbrply/pcap_util.py", line 120, in guess_parser
  File "/usr/local/lib/python3.8/dist-packages/usbrply-2.0.1-py3.8.egg/usbrply/pcap_util.py", line 78, in next
  File "/usr/local/lib/python3.8/dist-packages/pcapng/scanner.py", line 40, in __iter__
    yield self._read_next_block()
  File "/usr/local/lib/python3.8/dist-packages/pcapng/scanner.py", line 54, in _read_next_block
    raise ValueError('File not starting with a proper section header')
ValueError: File not starting with a proper section header
``
[ver.pcap.txt](https://github.com/JohnDMcMaster/usbrply/files/7263596/ver.pcap.txt)
`

JohnDMcMaster, 2021-10-01:
Thanks for the report! Will take a look soon

JohnDMcMaster, 2021-10-01:
If its in pcapng format, like yours, I can query the OS directly. Made a quick stab at it and pushed an update, let me know if this solves your issue

durka, 2021-10-01:
It runs now but it doesn't output any code. Did it work for you?

$ usbrply -p /tmp/ver.pcapng
# Generated by usbrply
# cmd: /usr/local/bin/usbrply -p /tmp/ver.pcapng

# Generated by usbrply
# Source: Linux pcap (usbmon)
# cmd: /usr/local/bin/usbrply -p /tmp/ver.pcapng
# PCapGen device hi: selected device 51
# WARNING: 1 pending complete requests
# PcapGen: generated 2 packets
# PcapGen device filter: dropped 0 / 2 packets
# SetupFilter: dropped 0 / 7 entries

durka, 2021-10-01:
I'm working with a device whose protocol is mostly undocumented/secret, and hoping to replicate whatever the official software is doing for a "change mode" command.

JohnDMcMaster, 2021-10-01:
If you open Wireshark you'll see there is only a single transaction which is not even complete. This is echoed in the output:

```
# WARNING: 1 pending complete requests
```

Did you select a portion to export from Wireshark? Please consider exporting everything instead and letting usbrply just keep the packets it needs

durka, 2021-10-02:
OK, I can try the capture again. I filtered the Wireshark display to only the outgoing packets before exporting.

JohnDMcMaster, 2021-10-02:
Gotcha. Ok I think your issue should be resolved now: I've improved OS detection and your capture is incomplete. But feel free to reach out again if you still have issues.

closed 2021-10-02 by JohnDMcMaster