JohnDMcMaster/usbrply #51: Seeing as python2 is EOL, consider changing pcap_util.py

humdogm, 2021-10-01:
Proposed changes:
Removing references to the pcap package and any library switching code.
Add explicit dependency on python-pcapng in setup.py

JohnDMcMaster, 2021-10-01:
Hmm would need to revisit, but some libraries were easier to setup, some where significantly more performant. I'd agree though that anything that is strictly python2 is probably ok dropping at this point. If that was only the python2 library maybe we unfortunately loose the high performance mode

humdogm, 2021-10-01:
Sorry, I actually was doing some more research, looks like somebody went and revived the [python-libpcap package](https://github.com/caizhengxin/python-libpcap), I was just going off the comment at the top of pcap_util.py

humdogm, 2021-10-01:
I see a couple alternatives on pypi, [python-libpcap](https://pypi.org/project/python-libpcap/) (different developer), and [libpcap](https://pypi.org/project/libpcap/)

JohnDMcMaster, 2021-10-01:
Heh getting a bit complicated now. At the end of the day nothing is broken and this looks like a bit of work to sort out properly. In theory if you were able to collect some performance numbers for the libraries it would help drive a decision. Or we say we no longer care about performance and just stick to the one library. I do a fair amount of camera reverse engineering though which is where the performant library was a huge plus.

JohnDMcMaster, 2021-10-01:
Aside: also curious if you can share more about your interest in the project / what you are working on. Are you working with durka?

humdogm, 2021-10-01:
Yeah, no problem. What I'm working on is related to this: https://github.com/bkerler/edl
I am trying to use the above tool to get write access to partitions on an Android phone without having to unlock its bootloader, in order to unbrick it without having to wipe its data.
I have managed to read all my device's partitions using it, but the userdata partition is encrypted. Also, the device manufacturer seems to have added some authentication mechanism which denies write access to the device's partitions.

However, the device manufacturer has their own proprietary and obfuscated tool which uses the same serial usb interface to communicate with the device and write data to its partitions, normally used to factory reset devices that cannot boot to other recovery modes.

My thought was that I could record the beginning of the manufacturers' partition writing sequence with USBPcap, replace the data with my own, and then replay the partition writing sequence.

closed 2021-10-02 by humdogm