JohnDMcMaster/usbrply #52: usbrply generates empty .py file

theGeekyLad, 2021-11-13:
I'm interested in bus-3 / device-50 which is my USB mouse:

```
$ lsusb

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0d62:3740 Darfon Electronics Corp. 
Bus 003 Device 004: ID 187c:0550 Alienware Corporation LED controller
Bus 003 Device 048: ID 050d:092a Belkin Components USB 2.0 BILLBOARD             
Bus 003 Device 047: ID 05e3:0751 Genesys Logic, Inc. microSD Card Reader
Bus 003 Device 044: ID 1a40:0801 Terminus Technology Inc. USB 2.0 Hub
Bus 003 Device 041: ID 050d:092b Belkin Components USB2.0 Hub             
Bus 003 Device 002: ID 0c45:6725 Microdia Integrated_Webcam_HD
Bus 003 Device 051: ID 06cb:0089 Synaptics, Inc. 
Bus 003 Device 050: ID 17ef:608d Lenovo Optical Mouse
Bus 003 Device 049: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 007: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I captured some packets through Wireshark and saved it as `test.pcapng` file followed by running:

```
usbrply --device-hi -p ~/test.pcapng > ~/t.py
```

The issue here is that the `t.py` file is empty and looks as follows:

```
# Generated by usbrply
# cmd: /usr/bin/usbrply --device-hi -p /home/thegeekylad/test.pcapng

# Generated by usbrply
# Source: Linux pcap (usbmon)
# cmd: /usr/bin/usbrply --device-hi -p /home/thegeekylad/test.pcapng
# PCapGen device hi: selected device 51
# PcapGen: generated 641 packets
# PcapGen device filter: dropped 632 / 641 packets
# SetupFilter: dropped 4 / 10 entries

```

I also tried:

```
usbrply --device 50 -p ~/test.pcapng > ~/t.py
```

i.e. by pointing it specifically to my device but couldn't find a way to pass the bus in question.

Kindly help me out!

JohnDMcMaster, 2021-11-13:
"--device 50" sounds reasonable for your use case given there are higher devices. you might also be able to use --vid/--pid (was broken at one point but IIRC fixed now). --device-hi will not work in your use case

Can you attach / link your pcap sample?

A few things to check:
-Wireshark only captures a single bus at a time. Did you capture the right bus? (3)
-Can you verify in Wireshark device 50 is present?
-Can you attach the output with "--device 50"

theGeekyLad, 2021-11-13:
Thanks for the prompt response!

Here's what I did in Wireshark:
[image]

And here's what I got when I ran `usbrply --device 50 -p ~/test.pcapng > ~/t.py`:
[usbrply_output.txt](https://github.com/JohnDMcMaster/usbrply/files/7531616/usbrply_output.txt)

Also, device 50, in my understanding, is actually present as it shows up in Wireshark (as `3.50.1`) and also in `lsusb`. Is there anything else I need to check?

JohnDMcMaster, 2021-11-13:
Looks like its roughly working to me: Wireshark shows interrupt reads and you have interrupts reads in your python output. Can you clarify what is not matching expectations? The only thing I'd add is that you could add --wrapper if you wanted a (relatively) complete program.

theGeekyLad, 2021-11-13:
Oh, but here's what happens when I execute that `t.py` file:

```
$ python ~/t.py 

Traceback (most recent call last):
  File "/home/thegeekylad/t.py", line 8, in <module>
    buff = interruptRead(0x81, 0x0004)
NameError: name 'interruptRead' is not defined
```

Am I missing something?

JohnDMcMaster, 2021-11-13:
Didn't get a chance to comment yesterday but I filed this: https://github.com/JohnDMcMaster/usbrply/issues/53

TLDR: you can fill in the VID/PID manually in the output as a workaround in the meantime. I have a fix half worked out but I'm short on time right now, so making "real" code changes is a bit harder

JohnDMcMaster, 2021-11-13:
Merged in a fix for above. Try updating and re-running

theGeekyLad, 2021-11-14:
Awesome, that was fast! It works now but I get another unrelated error which I'd have to scout around to fix:

```$ python ~/t.py 
Scanning for devices...

Found device
Bus 003 Device 010: ID 17ef:608d
Traceback (most recent call last):
  File "/home/thegeekylad/t.py", line 77, in <module>
    dev = open_dev(usbcontext)
  File "/home/thegeekylad/t.py", line 67, in open_dev
    return udev.open()
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 2056, in open
    mayRaiseUSBError(libusb1.libusb_open(self.device_p, byref(handle)))
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 128, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 120, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorAccess: LIBUSB_ERROR_ACCESS [-3]
```

Probably a `udev` rule should cut it but adding one didn't really help. Might just try logging out and in again and repost here. Thanks again!

theGeekyLad, 2021-11-14:
Okay, I fixed that with a `udev` rule followed by a reboot but now I see this:

```
$ python ~/t.py 

Scanning for devices...

Found device
Bus 003 Device 004: ID 17ef:608d
Traceback (most recent call last):
  File "/home/thegeekylad/t.py", line 78, in <module>
    dev.claimInterface(0)
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 1147, in claimInterface
    mayRaiseUSBError(
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 128, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/usr/lib/python3.9/site-packages/usb1/__init__.py", line 120, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorBusy: LIBUSB_ERROR_BUSY [-6]

```

theGeekyLad, 2021-11-14:
Was a bit too restless so with the help of the `usb1` `pydoc`, I modified `return udev.open()` in the generated Python code to the following:

```
usbDeviceHandle = udev.open()
usbDeviceHandle.detachKernelDriver(0)
return usbDeviceHandle
```

Here's the full program:
[usbrply-generated-py.txt](https://github.com/JohnDMcMaster/usbrply/files/7534451/usbrply-generated-py.txt)

Now there are no error messages and on executing it, I see this:
```
$ python ~/t.py 

Scanning for devices...

Found device
Bus 003 Device 015: ID 17ef:608d
```

But then I don't see the mouse move. 😄 Any suggestions?

JohnDMcMaster, 2021-11-14:
Glad to hear you are working through the issues :) Yes, you might run into permission and/or kernel issues, as you've noticed. Hmm maybe I'll add that note to the readme

However, I'm concerned with your overall project goal. You would only be able to send data to the mouse, not force the mouse to send data to the kernel. That is, you can do things like change led colors on the mouse (sent from the computer to the mouse) but not take mouse coordinates since those are sent from the mouse to the computer. It sounds like you want a Linux version of AutoHotkey?

theGeekyLad, 2021-11-16:
Sorry for the late reply but you precisely guessed my usecase! Nice @JohnDMcMaster :star: . I think your reasoning that it's kernel -> USB mouse and not the reverse is limpid and makes perfect sense now that I think of it. I basically need to get the LEDs on my built in laptop keyboard working on Linux (basically like building a control panel applet for Linux similar to what's on Windows).

That comes to my next question (which is partly answered on your README): can I _not_ virtual box Windows and instead, run Wireshark in a real Windows installation, capture the pcapng file and replay it inside Linux using this repo?

JohnDMcMaster, 2021-11-16:
That probably works: the current version of the software should support either Linux or Windows packet captures. Traditionally Linux has been more reliable though if I had to choose between them

JohnDMcMaster, 2021-11-18:
It sounds like your original question is answered. Please open a new issue though if you have new questions.

closed 2021-11-18 by JohnDMcMaster