JohnDMcMaster/usbrply #55: Doesn't work on Windows

theGeekyLad, 2021-11-23:
I followed the README closely but here's what happens on Windows:

### 1. `python -v`

Python 3.9.9

### 2. `python -m venv usbrply`

No errors. Here are more details:

```
PS C:\Users\rahul> ls usbrply

    Directory: C:\Users\rahul\usbrply

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----        23-11-2021     14:45                Include
d-----        23-11-2021     14:45                Lib
d-----        23-11-2021     14:45                Scripts
-a----        23-11-2021     14:45            164 pyvenv.cfg

```
```
PS C:\Users\rahul> ls .\usbrply\Scripts\

    Directory: C:\Users\rahul\usbrply\Scripts

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a----        23-11-2021     14:45           1970 activate
-a----        23-11-2021     14:45            962 activate.bat
-a----        23-11-2021     14:45          19408 Activate.ps1
-a----        23-11-2021     14:45            368 deactivate.bat
-a----        23-11-2021     14:45         106352 pip.exe
-a----        23-11-2021     14:45         106352 pip3.9.exe
-a----        23-11-2021     14:45         106352 pip3.exe
-a----        23-11-2021     14:45         542952 python.exe
-a----        23-11-2021     14:45         541928 pythonw.exe
```

### 3. `usbrply/Scripts/activate.bat`

Gives no output and I'm not in a Python virtual environment:

```
PS C:\Users\rahul> usbrply/Scripts/activate.bat
PS C:\Users\rahul>
```

### 4. `pip install usbrply`

Installs just fine:

```
PS C:\Users\rahul> pip install usbrply
Requirement already satisfied: usbrply in c:\users\rahul\appdata\local\packages\pythonsoftwarefoundation.python.3.9_qbz5n2kfra8p0\localcache\local-packages\python39\site-packages (2.1.1)
Requirement already satisfied: python-pcapng in c:\users\rahul\appdata\local\packages\pythonsoftwarefoundation.python.3.9_qbz5n2kfra8p0\localcache\local-packages\python39\site-packages (from usbrply) (1.0)
Requirement already satisfied: six in c:\users\rahul\appdata\local\packages\pythonsoftwarefoundation.python.3.9_qbz5n2kfra8p0\localcache\local-packages\python39\site-packages (from python-pcapng->usbrply) (1.16.0)
WARNING: You are using pip version 21.2.4; however, version 21.3.1 is available.
You should consider upgrading via the 'C:\Users\rahul\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\python.exe -m pip install --upgrade pip' command.
```

### 4. `python usrply\Scripts\usbrply -h`

Doesn't work. 😢 

```
PS C:\Users\rahul> python usrply\Scripts\usbrply -h
C:\Users\rahul\AppData\Local\Microsoft\WindowsApps\python.exe: can't open file 'C:\Users\rahul\usrply\Scripts\usbrply': [Errno 2] No such file or directory
```

Any suggestions?

JohnDMcMaster, 2021-11-23:
Thanks for the report! I'm not a Windows guy and didn't write these instructions, so might take me a bit to get back to you. But let me see if I can reproduce your issue in a VM

worldjoe, 2021-11-25:
Windows works for sure. I use it. The README could use some updating. I'll work on fork and submit some minor fixes.

worldjoe, 2021-11-25:
I would close this. rahul's step 4 has a typo that was present in the README before I fixed it.
4. python us**b**rply\Scripts\usbrply -h

JohnDMcMaster, 2021-11-26:
Thanks for the fix! Ah yeah I see that now. Sounds like should be resolved, but please re-open if otherwise

closed 2021-11-26 by JohnDMcMaster