JohnDMcMaster/usbrply #61: Interrupt data empty

himekifee, 2022-02-02:
```
Traceback (most recent call last):
  File "/home/grider/.local/bin/usbrply", line 139, in <module>
    main()
  File "/home/grider/.local/bin/usbrply", line 135, in main
    usbrply.printers.run(args.ofmt, filtered, argsj=argsj)
  File "/home/grider/.local/lib/python3.10/site-packages/usbrply/printers.py", line 14, in run
    printer.run(j)
  File "/home/grider/.local/lib/python3.10/site-packages/usbrply/pyprinter.py", line 232, in run
    self.parse_data(d)
  File "/home/grider/.local/lib/python3.10/site-packages/usbrply/pyprinter.py", line 208, in parse_data
    indented("interruptWrite(0x%02X, %s)" % (d["endp"], data_str))
UnboundLocalError: local variable 'data_str' referenced before assignment
```

JohnDMcMaster, 2022-02-03:
Thanks for the bug report! I'll see if I can reproduce this, but if you are able to upload your cap file it will help resolve it quicker. Thanks!

himekifee, 2022-02-03:
Thanks. Here is what I got.
[sc.zip](https://github.com/JohnDMcMaster/usbrply/files/7990824/sc.zip)

JohnDMcMaster, 2022-03-13:
Looking at this now. I think interruptWrite support was added and this was added as a theoretical during that but never tested. I'll use your pcap and see if I can come up with a patch.

JohnDMcMaster, 2022-03-13:
I've attempted a fix, but this hasn't seen real world testing so please let me know how it goes

closed 2022-03-13 by JohnDMcMaster