JohnDMcMaster/usbrply #62: Keyerror: device2vidpid on test script (and my own)

peddy22, 2022-03-11:
Hi, if I run the test script with `python usbrply\Scripts\usbrply --wrapper "C:\Users\Arno\Desktop\win1.pcapng"` I get the following error. Thanks for making the cool software and do you think you could help me out?

`    # Generated from packet 2213/2214
    buff = bulkRead(0x86, 0x0200)
    validate_read(b"\x08\xE6\x42\x02\x00\xB5\x40\x02\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
            b"\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x08\x00", buff, "packet 2213/2214")
    # PcapGen: generated 2476 packets
    # PcapGen device filter: dropped 609 / 2476 packets
    # SetupFilter: dropped 3 / 940 entries
Traceback (most recent call last):
  File "C:\Users\Arno\AppData\Local\Programs\Python\Python310\usbrply\Scripts\usbrply", line 139, in <module>
    main()
  File "C:\Users\Arno\AppData\Local\Programs\Python\Python310\usbrply\Scripts\usbrply", line 135, in main
    usbrply.printers.run(args.ofmt, filtered, argsj=argsj)
  File "C:\Users\Arno\AppData\Local\Programs\Python\Python310\usbrply\printers.py", line 14, in run
    printer.run(j)
  File "C:\Users\Arno\AppData\Local\Programs\Python\Python310\usbrply\pyprinter.py", line 236, in run
    if len(j["device2vidpid"]) != 1:
KeyError: 'device2vidpid'`

JohnDMcMaster, 2022-03-11:
Thanks! Taking a look

By coincidence I finally got a chance to work on the backlog of issues this week and there is some outstanding code on my laptop that fixes a number of bugs and greatly expands the test suite. I'll check if these changes fix this

JohnDMcMaster, 2022-03-13:
Alright pushed out some significant changes which I think improves but not fixes this ("Exception: Failed to guess vid/pid: found 0 device entries"). Let me look into this. However, if you are able to post your .pcap file it might help expedite the fix

peddy22, 2022-03-13:
Thanks John - I don't think I was having this issue, just KeyError:
'device2vidpid'`  That was shown in the original issue post. I'm sure if it
works in the test pcap it'll work on mine too. I'll try it later.

JohnDMcMaster, 2022-03-13:
I understand its not your exact issue, but I'm saying I found something that might be related.

In any are you able to post your pcap file? It might help me solve it quicker

JohnDMcMaster, 2022-03-14:
TLDR: are you running the latest software from git? Several things suggest you might have an older release (pip?). If you can confirm the latest git fixes I'll cut a new pip release

Details:

First, the other issue has been fixed. It sounds like it is something different, but it would have caused you issues soon. Here if you are curious: https://github.com/JohnDMcMaster/usbrply/pull/68

Anyway, a few things from your output:
-I now see that win1.pcapng was the old name of a file in the test suite. This is now called win_misc.pcapng
-Are you running the latest version from git? (not pip)
-Aside: you are running "python" not "python3". I'll assume "python" is Python 3 as I'm not sure this even runs on Python 2 anymore

The specific reason for your crash appears to be that VidpidFilter filter never ran, so it didn't create a vidpid map. You can see this missing filter in the comments in the output. It is hard coded into main.py, so I'm not sure why it wouldn't run if you have the latest software (https://github.com/JohnDMcMaster/usbrply/blob/master/usbrply/main.py#L125)

Here is your output:
```
python usbrply\Scripts\usbrply --wrapper "C:\Users\Arno\Desktop\win1.pcapng"
...
    # PcapGen: generated 2476 packets
    # PcapGen device filter: dropped 609 / 2476 packets
    # SetupFilter: dropped 3 / 940 entries
```
Here is what I see:
```
$ usbrply --wrapper test/data/win_misc.pcapng
...
    # PcapGen: generated 2476 packets
    # PcapGen device filter: dropped 609 / 2476 packets
    # VidpidFilter: dropped 0 / 940 entries, want None
    # SetupFilter: dropped 3 / 941 entries
...
```

peddy22, 2022-03-14:
Definitely not running the latest git version, I'll get it now

peddy22, 2022-03-14:
So to update, the latest update fixes the error mentioned in the original post of this issue. Thanks for the software!

Side note- I need to learn a bit more about the wonderful USB protocol before I attempt to writer driver software for my RGB keyboard. 

Cheers

closed 2022-03-14 by peddy22

JohnDMcMaster, 2022-03-14:
Huzzah! Thanks for the feedback

Good luck on your project. Feel free to ping me if you get stuck at some point and/or need some pointers to get started