JohnDMcMaster/usbrply #70: error on packet

Jefferson-Lopes, 2022-04-05:
My setup: I'm reading the USB packet of a windows machine that's communicating with the Pegasus Amatrol robotics arm. This arm has an FTDI FT2232D chip and the driver used is FTDI version 2.8.28.0. I already tried to use the WinUSB driver but it didn't work. I'm running the USBrply on Ubuntu 20.04 LTS base.

Problem: the USBrply can't understand packet 279 of the 04-04-22_1.pcapng file, but I'm sure that this record is of valid data transmission.

When seeing packet 279 on Wireshark, the only thing that I notice is IRP USBD_STATUS: Unknown (0x00000078). For comparison, in packet 149, which have the same function URB Function: URB_FUNCTION_GET_DESCRIPTOR_FROM_DEVICE (0x000b) there is: IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)

On the python code created, I have only the message "oh noes!
ERROR: packet 279"

Command used to run the USBrply: usbrply --wrapper --verbose --device-hi -p 04-04-22_1.pcapng

04-04-22_1.pcapng file: [04-04-22_1.pcapng](https://drive.google.com/file/d/1I40orgW6nKK-QUFWnyPTWQiZ71r8msKk/view?usp=sharing)

JohnDMcMaster, 2022-04-05:
Thanks for your report! I'm looking over your pcap file and will get back to you

JohnDMcMaster, 2022-04-20:
Looks like it might be a USB PHY error. I think this was some very very old placeholder error handling code and since I've never seen a PHY error I never had a test case against it. Going to do a few things...
1) Add this to the test suite so its more explicitly handled
2) improve the error handling (better message and/or don't sys.exit())
3) Probably change the behavior to drop the packet and print a warning

JohnDMcMaster, 2022-04-20:
Also you can probably work around this by specifying --no-halt. For example:

```
$ usbrply --no-halt gh-70_04-04-22_1.pcapng
```

Runs to completion with a few warmings:

```
# WARNING: 2 pending complete requests
# PcapGen: generated 14069 packets
# PcapGen device filter: dropped 0 / 14069 packets
# VidpidFilter: dropped 0 / 7042 entries, want None
# SetupFilter: dropped 10 / 7043 entries
```

JohnDMcMaster, 2022-04-20:
Alright looking through SDK documentation / headers (https://github.com/tpn/winsdk-10/blob/master/Include/10.0.14393.0/shared/usb.h), it looks like this is technically ok:

```
#define USBD_ERROR(Status) ((USBD_STATUS)(Status) < 0)
```

So need to do the following:
-Fix struct definition to be signed
-Change error detection
-Since this is unusual though, I might omit a warning if verbose is specified

Jefferson-Lopes, 2022-04-21:
Thanks for your reply. I was using the ```--no-halt``` flag and it worked perfectly.

Another problem, though, is that I can't connect to my FTDI device as soon as it's plugged into my Linux PC. My workaround was to run a script to detach the interfaces 0 and 1 from the kernel. I'm not sure if that's a problem with my device in particular, but I'm hoping you can give some insight.

```python
import usb1

VID = 0x0403
PID = 0x6010

with usb1.USBContext() as context:
    handle = context.openByVendorIDAndProductID(
        VID,
        PID,
        skip_on_error=True,
    )

    if handle is None:
        print('Device not present, or user is not allowed to access device')
        exit
    else:
        print('Connected to 0x%04x:0x%04x device'%(VID, PID))

        print('\nBefore detach interfaces:')

        for i in range(5):
            kernelDriver = handle.kernelDriverActive(i)
            if kernelDriver:
                handle.detachKernelDriver(i)
            print(F'  Kernel driver interface {i}: ' + ('activated' if kernelDriver else 'deactivated'))

        print('\nAfter detach interfaces:')

        for i in range(5):
            kernelDriver = handle.kernelDriverActive(i)
            print(F'  Kernel driver interface {i}: ' + ('activated' if kernelDriver else 'deactivated'))
```

closed 2022-04-21 by JohnDMcMaster

JohnDMcMaster, 2022-04-21:
Merged a fix that should be more proper

Re: kernel claiming device. hmm you have a few options but that's probably the best. You could also unload/blacklist the ftdi kernel module if you wanted a bigger hammer

Jefferson-Lopes, 2022-04-21:
It's good to know that this is a proper solution.  Again, thanks for your help.

PR #72 (2022-04-21): Gh70
Fixes #70 and #71

I've significantly improved the handling of Windows special packets. I've also added a few test cases with some sample output below

```
$ usbrply test/data/win_abort-pipe.pcapng
...
# Generated from packet 1/2
# ABORT_PIPE()

$ usbrply test/data/win_pipe-stall.pcapng
...
# Generated from packet 1/2
# IRP_INFO(): func SYNC_RESET_PIPE_AND_CLEAR_STALL
```