JohnDMcMaster/usbrply #75: Fails while parsing URB_INTERRUPT out

sbonnick, 2022-05-17:
Hi. Just started using your tool to see if it will work for me. ran into a issue "ERROR: packet 8". I belive this corresponds to the 8th frame in wireshark,  I have attached my wireshark log output.   I'm using this in windows.

sbonnick, 2022-05-17:
I am still learning more about USB communication, but Looking at this thread, I wonder if the problem is in Wireshark-windows all URB info is "URB_INTERRUPT" with either "in" or "out" direction. where on Linux it may show up as "USB_SUBMIT" or "USB_COMPLETE"  

https://superuser.com/questions/873896/wireshark-usb-traces-explanations/876773#876773

JohnDMcMaster, 2022-05-17:
Thank you for attaching sample data. I'll take a look soon

Yes, they are different, but Windows and Linux is handled separately and in theory should account for this. However interrupt support in general isn't well tested. I'll look over your data and add it to the test suite as appropriate

sbonnick, 2022-05-18:
Awesome Thanks.

Here is also the error log. 

```
python C:\Users\NAME\AppData\Local\Programs\Python\Python310\Scripts\usbrply -p test.pcapng >replay.py
Traceback (most recent call last):
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\Scripts\usbrply", line 138, in <module>
    main()
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\Scripts\usbrply", line 135, in main
    usbrply.printers.run(args.ofmt, filtered, argsj=argsj)
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\lib\site-packages\usbrply\printers.py", line 14, in run
    printer.run(j)
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\lib\site-packages\usbrply\pyprinter.py", line 231, in run
    for d in j["data"]:
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\lib\site-packages\usbrply\setup_filter.py", line 31, in gen_data
    for data in datas:
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\lib\site-packages\usbrply\com_pcap.py", line 82, in gen_data
    if not parser.next(self.loop_cb):
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\lib\site-packages\usbrply\pcap_util.py", line 84, in next
    loop_cb(block.captured_len, block.packet_data, block.timestamp)
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\lib\site-packages\usbrply\win_pcap.py", line 452, in loop_cb
    self.process_complete(dat_cur)
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\lib\site-packages\usbrply\win_pcap.py", line 514, in process_complete
    self.processInterruptCompleteOut(dat_cur)
  File "C:\Users\NAME\AppData\Local\Programs\Python\Python310\lib\site-packages\usbrply\win_pcap.py", line 701, in processInterruptCompleteOut
    assert self.submit.m_urb.data_length == 0
AssertionError
```

JohnDMcMaster, 2022-05-18:
ah ok this was known to maybe be problematic:

```
    def processBulkCompleteIn(self, dat_cur):
        # looks like maybe windows doesn't report the request size?
        # think this is always 0
        assert self.submit.urb.data_length == 0
```

anyway will take a look

semarainc, 2022-11-01:
is there any progress regarding this issue?

thanks :D