JohnDMcMaster/usbrply #77: Asnyc BULK parsing error

jwinarske, 2023-02-28:
I have a file that fails to parse with output looking like this:
```
# PCapGen device hi: selected device 0
# WARNING: Packet 3 missing submit.  URB ID: 0x0000000000120028
# WARNING: Packet 4 missing submit.  URB ID: 0x0000000000120128
# WARNING: Packet 7 missing submit.  URB ID: 0x0000000000120028
# WARNING: Packet 8 missing submit.  URB ID: 0x0000000000120128
# WARNING: Packet 11 missing submit.  URB ID: 0x00000000000E0028
# WARNING: Packet 12 missing submit.  URB ID: 0x00000000000E0128
# WARNING: Packet 15 missing submit.  URB ID: 0x0000000000180028
# WARNING: Packet 16 missing submit.  URB ID: 0x0000000000180128
# WARNING: Packet 19 missing submit.  URB ID: 0x00000000002E0028
# WARNING: Packet 20 missing submit.  URB ID: 0x00000000002E0128
# WARNING: Packet 23 missing submit.  URB ID: 0x0000000000010028
# WARNING: Packet 24 missing submit.  URB ID: 0x0000000000010128
# WARNING: Packet 26 missing submit.  URB ID: 0x0000000000FF0028
# WARNING: Packet 28 missing submit.  URB ID: 0x0000000000040128
# WARNING: Packet 31 missing submit.  URB ID: 0x0000000002040028
# WARNING: Packet 32 missing submit.  URB ID: 0x0000000000320128
# WARNING: Packet 35 missing submit.  URB ID: 0x0000000002040028
# WARNING: Packet 36 missing submit.  URB ID: 0x0000000000220128
# WARNING: Packet 39 missing submit.  URB ID: 0x0000000002040028
# WARNING: Packet 40 missing submit.  URB ID: 0x0000000000560128
oh noes!
ERROR: packet 42
```

jwinarske, 2023-03-08:
This is on Linux.

Basically 10 bulk reads get scheduled per endpoint, which complete later (async).  This pattern continues with 10 reads always in flight.  It's used to maximize throughout.

raleighlittles, 2023-03-25:
@jwinarske how did you run this tool? I'm also on Linux (Ubuntu 20) and the instructions didn't seem to work. I did:

* `sudo pip install python-pcapng` to install (from here: https://github.com/JohnDMcMaster/usbrply#linux-installation)
*  `usbrply --wraper --device-hi -p ~/github/ITOSAYDE-MSR-880/wireshark/magnetic-card-read-only-read-button_1-34.pcapng` (from here: https://github.com/JohnDMcMaster/usbrply#sample-workflows)

But I get this error: `zsh: command not found: usbrply`