JohnDMcMaster/usbrply #79: Failed to guess vid/pid

ajmeese7, 2023-07-28:
I got the following errors when I tried to run `usbrply --wrapper --device-hi -p Stolen_flights.pcapng > replay.py`:

```log
Traceback (most recent call last):
  File "/usr/local/bin/usbrply", line 33, in <module>
    sys.exit(load_entry_point('usbrply==2.1.1', 'console_scripts', 'usbrply')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/usbrply-2.1.1-py3.11.egg/usbrply/main.py", line 140, in main
  File "/usr/local/lib/python3.11/dist-packages/usbrply-2.1.1-py3.11.egg/usbrply/printers.py", line 17, in run
  File "/usr/local/lib/python3.11/dist-packages/usbrply-2.1.1-py3.11.egg/usbrply/pyprinter.py", line 246, in run
Exception: Failed to guess vid/pid: found 0 device entries
```

This is the [PCAP file](https://drive.google.com/file/d/1JXvykSw9X2r3DDibHlxFJh08hbNUoGwo/view?usp=sharing) in question, I'm still very inexperienced working with USB data captures so this may be "expected" behavior here. If it is, some error handling for this use case would be useful!

Very neat tool, thank you for sharing.

bsaleh03, 2023-09-19:
Any update on this issue? I'm facing the same one.

JohnDMcMaster, 2023-09-29:
Unfortunately not. There is a project that was supposed to use usbrply that's been pushed back a bunch of times. Still haven't gotten to it and pretty busy with other work.